JdeRobot/VisualStates #1: Migration from the JdeRobot/JdeRobot repository

jmplaza, 2018-02-28:
VisualStates will have its own standalone repository. This will make easier to generate a ROS package for it, the development of this tool and the creation of an AppImage of it.
Right now the source code of the tool and its examples are inside the JdeRobot/JdeRobot repository.

okanasik, 2018-03-15:
Dear @jmplaza,
I created a separate folder for visualstates, but I do not have permission to push here. Will you give me permission? or Do I need to create a pull request. Best.

jmplaza, 2018-03-15:
Sure @okanasik, done, you can push now. Anyway maybe it is a better policy to create a pull request and then you can merge it yourself. This way all the changes go through the issue + pull_request mechanism. It is a better policy in the long run. It makes easier to track the development and the double check by other developer before merging. Let's follow it, please.

okanasik, 2018-03-15:
Dear @jmplaza, as suggested I created a pull request.

closed 2018-03-19 by okanasik