JdeRobot/VisualStates #11: [Deps] Python dependecies in CMakeLists.txt

pushkalkatara, 2018-04-07:
The CMakeLists.txt is currently lacking following Python related dependencies - 
sysv_ipc==1.0.0
PyYAML==3.12
PyQt5 needs to be added as a dependency but as further updates in PyQt5, its support for [Python2](http://pyqt.sourceforge.net/Docs/PyQt5/installation.html#installing-from-wheels) is removed. Also while installing pyqt5 using `sudo apt-get install python-pyqt5`, package PyQt5.Qsci is missing. To solve this would require installing PyQt4 and PyQt5 alongside. How to solve this issue?

okanasik, 2018-04-16:
Dear @pushkalkatara,

The Ubuntu 16.04 provides packages for PyQt5, and PyQt5.QSci for python2.7. However, I am not sure whether it is necessary to put these dependencies in cmake. What do you think @reysam93 @jmplaza

jmplaza, 2018-04-16:
Maybe we can put the dependences on those Ubuntu-16.04 packages (PyQt5, and PyQt5.QSci for python2.7), as long as they both are already required to work and available. It doesn't hurt, they are going to be easily satisfied and installed in case of using VisualStates from Ubuntu 16.04. Aren't they? 

We will see for the upcoming Ubuntu-18.04. 

We have a serious problem with Python in JdeRobot, and so, with VisualStates: I would like to jump to Python-3.X ASAP, but in order to keep compatibility with ROS packages we have to stick to Python-2.7. Do you know a way to communicate a Python3 node with a Python-2.7 ROS driver node?

okanasik, 2019-02-04:
We resolved that issue since the dependencies to sysv_ipc and PyYAML are removed.

closed 2019-02-04 by okanasik