JdeRobot/VisualStates #17: [BUG] Remove state option does not update TreeModel

pushkalkatara, 2018-05-28:
While removing a state from automata scene by, the state disappears from the automata scene but not from the tree model.

okanasik, 2018-05-30:
Dear @pushkalkatara, can you also write the steps to reproduce this error. I am not able to reproduce while creating a automata. Best,

pushkalkatara, 2018-05-30:
Respected Sir,
When removing a substate of a state, the state gets removed from the automata scene but not from the TreeView like in the case below.
[image]

But in the case when a state is removed from the root state, the TreeView is also updated like in the case below.
[image]

PR #19 (2018-05-30): Fixing subautomata removal bug
Fixes #17 by referencing the parent state from the treemodel.
closed 2018-05-30 by okanasik