JdeRobot/VisualStates #22: Behavior Data(Configs, Functions, Aux Variables) while importing a state.

pushkalkatara, 2018-06-04:
In the case when a behavior is being imported, the importing state would have some configurations, functions and aux variables. to generate file ros nodes, these configurations, functions and aux variables must exist in the developing state.
Configurations can be appended to the configurations list while checking if it exists before, but functions and auxiliary variables must be unique, as there may be name conflict etc. Please share views on this issue. @jmplaza @okanasik

okanasik, 2018-06-05:
I think the simplest strategy to solve this problem would be concatenating a string to the imported states. According to this name change, we also need to update all names used in the functions and variables. We can also ask to the user for a unique name for the imported function and variables.
Another strategy would be encapsulating all functions and variables in different namespace or class. Therefore, when user would like to edit functions or variables, it chooses a namespace and also access these functions and variables through same namespace.

pushkalkatara, 2018-06-08:
Namespace and Class Strategy:
* Adding a new tag in state and tranisition xml parent tag defining the class or namespace to use.
* Functions and variables parsed and appended to the dictionaries of namespaces. Ex - Every namespace would be a unique key in dictionary, and a sub dictionary containing all the functions.
* While generating code, a new object would be created, and the imported functions and variables would be prefix with the new namespace to use that class's methods and variables. 
Pros:
* Need to just prefix the code and transition code with the class name. i.e self.className.(function or variable)
Cons:
* User would need to choose namespaces or classes name while writing functions and variables in the dialogs. 

Name Mangling Strategy:
* Imported functions and variables would be parsed with importID which keeps iterating on every import. I came accross a realted stackoverflow post - [https://stackoverflow.com/questions/11024344/python-name-mangling-function](https://stackoverflow.com/questions/11024344/python-name-mangling-function)
Pros:
* No need for user to choose the namespace. Just write code in state, transition and variables with the defined names.
Cons:
*  The code functions and variables in the state and transition must be updated with the new names according to the name mangling output.
@jmplaza @okanasik Please guide me over the strategy.

Basically making classes in Python is overhead namemangling only. So a low level implementation would be the mangling strategy. 
Also Python provides a way to perform name mangling. I am not sure this may be useful for our usecase, just thought worth mentioning - 
Double Leading Underscore __var triggers name mangling when using in a class context. Enforced by Python interpreter. Applies to variables and methods. Ex - 
```
class Test:
    def __init__(self):
        self.foo = 11
        self._bar = 23
        self.__baz = 23

Output - 
>>> t = Test()
>>> dir(t)
['_Test__baz', '__class__', '__delattr__', '__dict__', '__dir__',
 '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__',
 '__gt__', '__hash__', '__init__', '__le__', '__lt__', '__module__',
 '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__',
 '__setattr__', '__sizeof__', '__str__', '__subclasshook__',
 '__weakref__', '_bar', 'foo']
```
Python interpreter created a variable _Test__baz appending the class name to the variable, instead of making it globally available.
Hence while using it globally -
```
>>> t.__baz
AttributeError: "'ExtendedTest' object has no attribute '__baz'"
>>> t._Test__baz
23
```
Could this help us in some way?

pushkalkatara, 2018-06-29:
Name mangling under work, Import issue solved with the manual merge.

closed 2018-06-29 by pushkalkatara