JdeRobot/VisualStates #35: Name Mangling while importing functions and variables.

pushkalkatara, 2018-06-29:
Namespace and Class Strategy:
* Adding a new tag in state and tranisition xml parent tag defining the class or namespace to use.
* Functions and variables parsed and appended to the dictionaries of namespaces. Ex - Every namespace would be a unique key in dictionary, and a sub dictionary containing all the functions.
* While generating code, a new object would be created, and the imported functions and variables would be prefix with the new namespace to use that class's methods and variables. 
Pros:
* Need to just prefix the code and transition code with the class name. i.e self.className.(function or variable)
Cons:
* User would need to choose namespaces or classes name while writing functions and variables in the dialogs. 

Name Mangling Strategy:
* Imported functions and variables would be parsed with importID which keeps iterating on every import. I came accross a realted stackoverflow post - [https://stackoverflow.com/questions/11024344/python-name-mangling-function](https://stackoverflow.com/questions/11024344/python-name-mangling-function)
Pros:
* No need for user to choose the namespace. Just write code in state, transition and variables with the defined names.
Cons:
*  The code functions and variables in the state and transition must be updated with the new names according to the name mangling output.
@jmplaza @okanasik Please guide me over the strategy.

Basically making classes in Python is overhead namemangling only. So a low level implementation would be the mangling strategy. 
Also Python provides a way to perform name mangling. I am not sure this may be useful for our usecase, just thought worth mentioning - 
Double Leading Underscore __var triggers name mangling when using in a class context. Enforced by Python interpreter. Applies to variables and methods. Ex - 
```
class Test:
    def __init__(self):
        self.foo = 11
        self._bar = 23
        self.__baz = 23

Output - 
>>> t = Test()
>>> dir(t)
['_Test__baz', '__class__', '__delattr__', '__dict__', '__dir__',
 '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__',
 '__gt__', '__hash__', '__init__', '__le__', '__lt__', '__module__',
 '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__',
 '__setattr__', '__sizeof__', '__str__', '__subclasshook__',
 '__weakref__', '_bar', 'foo']
```
Python interpreter created a variable _Test__baz appending the class name to the variable, instead of making it globally available.
Hence while using it globally -
```
>>> t.__baz
AttributeError: "'ExtendedTest' object has no attribute '__baz'"
>>> t._Test__baz
23
```
Could this help us in some way?

okanasik, 2018-06-29:
Dear @pushkalkatara , I think the name mangling approach is not user friendly. It is better to follow namespace approach. Maybe in the function and variables dialog we can have an option to choose the namespace our function and variables will reside. However, I think it is better to follow the simplest approach. If there is a conflict about the names of the variables and functions. We should show a dialog to the user to rename whatever he wants. I think that would be more user friendly. Then, we can investigate namespace approach.

pushkalkatara, 2018-06-30:
I'll try and demo the strategies with an example:
Currently in Kobuki Example of obstacleAvoidance:
There exists a class RosNode which contains the ROS related publishers and subscribers variables and callbacks. With these, it also contains the user-defined methods and variables. I noticed that state code only requires a publisher related methods and variables and it doesn't require user created aux data. Only the transition class requires the user defined data to check a condition. Would there be any case in which a state would require the user aux data?
We can make an update to the existing code if feasible -
```
class RosNode():
	def __init__(self):
		rospy.init_node("test", anonymous=True)

		self.turtlebotROS_mobile_base_commands_velocityPub = rospy.Publisher("/turtlebotROS/mobile_base/commands/velocity", Twist, queue_size=10)
		self.turtlebotROS_laser_scanSub = rospy.Subscriber("/turtlebotROS/laser/scan", LaserScan, self.turtlebotROS_laser_scanCallback)
		self.turtlebotROS_laser_scan = LaserScan()

		time.sleep(1) # wait for initialization of the node, subscriber, and publisher

	def stop(self):
		rospy.signal_shutdown("exit ROS node")

	def publishturtlebotROS_mobile_base_commands_velocity(self, turtlebotROS_mobile_base_commands_velocity):
		self.turtlebotROS_mobile_base_commands_velocityPub.publish(turtlebotROS_mobile_base_commands_velocity)

	def turtlebotROS_laser_scanCallback(self, turtlebotROS_laser_scan):
		self.turtlebotROS_laser_scan = turtlebotROS_laser_scan

class Namespace1():
	def __init__(self, rosNode):
		self.obstacle_threshold = 0.4
		self.is_obstacle = False
		self.rosNode = rosNode
	
	def calculate_obstacle(self):
		laserData = self.rosNode.turtlebotROS_laser_scan
		for val in laserData.ranges:
			if val < self.obstacle_threshold:
				self.is_obstacle = True
				return
		self.is_obstacle = False
```
And in the transitions
```
class Tran1(ConditionalTransition):
	def __init__(self, id, destinationId, namespace):
		ConditionalTransition.__init__(self, id, destinationId)
		self.namespace = namespace

	def checkCondition(self):
		self.namespace.calculate_obstacle();
		return self.namespace.is_obstacle;

	def runCode(self):
		pass
```
in the main code
```
if __name__ == "__main__":
	rosNode = RosNode()
	namespace0 = Namespace0(rosNode)
        tran1 = Tran1(1, 2, namespace0)
	state1.addTransition(tran1)
```

okanasik, 2018-07-01:
Dear @pushkalkatara,
I think this namespace approach sounds fine. However, all transitions and state should be able to access all namespaces and rosnode functionality. Therefore, we can pass a rosNode and another class object that have a reference all namespaces (like namespace container). We can assume that if no state is imported we have a root namespace in our namespace container. When a state machine is imported we replace reference to root namespace of that state with the name of the state. I think that would be cleaner approach, but if you find better approach we can discuss that also.
Best,

pushkalkatara, 2018-07-02:
I'll quick implement this approach. We'll find out further difficulties with an example. :D
Thanks for the update.