JdeRobot/VisualStates #49: XML parsing and generating for Namespace functionality

pushkalkatara, 2018-07-03:


pushkalkatara, 2018-07-29:
With recent commit generating namespaces is resolved but parsing still remains.

pushkalkatara, 2018-07-29:
Regarding parsing of namespaces, we can implement two strategies - 
1. Adding a new XML tag in every state which directly tells the ID of the namespace. This would be an easy strategy but not efficient.
2. As parsing of states and transitions is through a depth-first approach, it would be inefficient to find the namespace level again and again while applying it to states and transitions. Workaround to this is a breadth-first search implementation. Using it, on every level an object of namespace would be formed once and would be assigned to all the states and transitions parsing them altogether.
Please suggest @okanasik @jmplaza

pushkalkatara, 2018-07-30:
Here's a quick implementation on the branch [parsing-namespace](https://github.com/pushkalkatara/VisualStates/tree/parsing-namespace). In the fileManager file there needs to be an array of namespaces which contains all the namespaces. But with the current implementation, there is no proper way to extract all the namespaces from each state level.
Also as rootState would not have any Namespace, it is creating an issue with parsing.

okanasik, 2018-07-31:
Dear @pushkalkatara, I am a little confused about that issue. Every state is supposed to have single `local_namespace`. When you import **in** or **on** a state, you can directly access `local_namespace` of those states and merge them. That is it. You do not need to parse whole state tree for every import action. Even we do not need to have id or name of a `local_namespace` since every `local_namespace` is associated with exactly one unique state.

When I check the implementation, the problem would be due to keeping all namespaces as a list. Actually, you do not need to keep separate list of namespaces. Since you have states, you can directly access `local_namespace` of those states without iterating over namespaces. When the user wants to edit functions and variables, you should get the active state, get the `local_namespace` of that active state and let the user edit that. If I miss something or you have questions please write.

pushkalkatara, 2018-08-01:
Sir @okanasik, the current implementation is for open functionality when a user opens a new automaton through an XML file. In this case, multiple levels are being parsed which would have different states and namespaces. Also, states at the same level would have the same local namespace.
As the states are being parsed depth-wise, list for namespaces is required, but if we change the implementation to breadth-wise parsing, we can create an object of the namespace and assign it to all states at that level, without appending the namespace to the list. Also to generate namespaces in the current implementation, the namespace list is being used. But we can extract all the namespaces with the initial child also. Yes the list is not required, but the parsing strategy must be changed for implementation.

okanasik, 2018-08-02:
Dear @pushkalkatara, the states at the same level would have an access to their parents local namespace. Actually namespaces will be applied only to the states having one or more child states. These child states can access anything defined on the local namespace of their parents. Therefore, it should not matter how you parse your tree. When you select a state on the treeview, you actually choose a state and on the canvas you see their children. When I would like to change functions and variables, I will be able to change local namespace of the selected state. Maybe you are confused about the depth level local namespace, but finally we decided to implement local namespace specific to the states and children of that state can access that local namespace.

closed 2018-08-12 by pushkalkatara