JdeRobot/VisualStates #60: None while displaying Global and Local Namespace variables.

pushkalkatara, 2018-08-13:
This issue might be due to the delay in ipc library. @okanasik please review.

XML File to reproduce the function - 
```
<?xml version="1.0" ?>
<VisualStates>
  <config type="1">
    <buildDependencies/>
    <runDependencies/>
    <topics/>
  </config>
  <global_namespace>
    <functions>def helloWorld(self):
	print(&quot;Hello World&quot;)

</functions>
    <variables>self.variable = 0</variables>
  </global_namespace>
  <libraries/>
  <state id="0" initial="True">
    <posx>0.0</posx>
    <posy>0.0</posy>
    <name>root</name>
    <code></code>
    <timestep>100</timestep>
    <namespace>
      <functions>def helloFromState1(self):
	print(&quot;Hello From State 1&quot;)
	
</functions>
      <variables>self.state1variable = 0</variables>
    </namespace>
    <state id="1" initial="True">
      <posx>959.0</posx>
      <posy>863.0</posy>
      <name>state1</name>
      <code>print(self.namespace.helloFromState1())
print(self.globalNamespace.helloWorld())</code>
      <timestep>100</timestep>
      <namespace>
        <functions></functions>
        <variables></variables>
      </namespace>
    </state>
  </state>
</VisualStates>
```

[image]

okanasik, 2018-08-14:
I have tested with the latest state from the branch `parsing-namespace`. There is a small error in the order of the parameters, namespace is initialized with 100 (cycleDuration). Can you check this error?

What is the error in this video? I could not get the error. Can you elaborate more?

okanasik, 2018-08-15:
Fixed.

closed 2018-08-15 by okanasik