JdeRobot/VisualStates #71: Connecting VisualStates with Pepper robot.

fqez, 2018-11-26:
Hi guys,
I've managed to connect the Pepper robot from Softbank robotics to the VisualStates tool, as you can see [here!!](https://www.youtube.com/watch?v=vUFg1aTA7-k).
**Congratulations** for such an amazing tool which opens a new world of possibilities for programming autonomous behaviours for robots in an easy way.
I almost had no problem doing this test, but the way the publisher and suscribers variables are named (things like "publishname_of_the_topic(...)" seems kinda counterintuitive. Nonetheless the tool works really well and I find it so powerful.
**CONGRATULATIONS!** :)

okanasik, 2018-11-27:
@fqez Thank you for your nice words. Yes, we can to improve our UI to let the developer give a specific function names for subscribed data and publish methods.

okanasik, 2018-11-27:
I will create a new feature request issue to improve config UI.

closed 2018-11-27 by okanasik