JdeRobot/VisualStates #91: [enhancement] Interface to select states to import

pushkalkatara, 2019-03-03:
Currently, using Import functionality, all the states are imported in the developing behavior. In some cases, only some of the selected states are required, hence an interface which display all the states and provides functionality to select one or all states to be imported.

sudo-panda, 2019-04-11:
Does the interface show the visual of transitions with the states or only the state names?

pushkalkatara, 2019-04-11:
Hi, @sudo-panda Thanks for approaching the issue, we can start initially just with displaying the states.

sudo-panda, 2019-04-16:
Are the state ids always continuous or can there be gaps and does the child states have to have greater state ids than the ones in the upper level?

pushkalkatara, 2019-04-17:
Hi @sudo-panda , the child states need not be continuous, also it's not necessary that the child state would have greater state ids compared to the ones in the upper level. The whole procedure works on recursive parsing.

sudo-panda, 2019-04-17:
Shouldn't a deleted state delete a destination Transitions along with Origin Transition both during  importing and in the main VisualStates window. Right now it only deletes origin Transitions.

pushkalkatara, 2019-04-17:
I didn't get you, what do you mean by origin transition or destination
transition? I believe a transition points from origin state to a
destination state. If we remove a particular state, all the transitions
attached to it are automatically removed. But if a transition is removed,
both the states connected to it stays in the automata.

sudo-panda, 2019-04-17:
That should be the ideal behaviour but right now only the transitions that originate from a state are removed on removal of the state and not the ones that point towards the state

sudo-panda, 2019-04-20:
Should I modify the current behaviour?

sudo-panda, 2019-04-25:
@pushkalkatara could you please reply so that finish adding this feature.

pushkalkatara, 2019-04-28:
Hi @sudo-panda . I was busy with something. I think we would require more discussion over this, it would be better to create a separate issue for it.

sudo-panda, 2019-04-28:
No problem. Should I create a pull request to add this feature or first solve that issue?

pushkalkatara, 2019-04-28:
Both are unrelated, you can proceed with any of them.

pushkalkatara, 2019-08-10:
Developing as part of GSoC 2k19 at https://github.com/TheRoboticsClub/colab-gsoc2019-Baidyanath_Kundu/pulls

closed 2019-08-10 by pushkalkatara