JdeRobot/VisualStates #92: [good-first-issue] Transition takes empty name as string but fails to load.

pushkalkatara, 2019-03-03:
While developing behavior, transitions take the empty name as input. After saving the behavior and reopening it, the following error occurs - 
```
Traceback (most recent call last):
  File "/home/pushkalkatara/catkin_ws/src/VisualStates/scripts/main.py", line 30, in <module>
    mainWindow.openFile(sys.argv[1])
  File "/home/pushkalkatara/catkin_ws/src/VisualStates/src/visualstates/gui/visualstates.py", line 215, in openFile
    (self.rootState, self.config, self.libraries, self.globalNamespace) = self.fileManager.open(fileName)
  File "/home/pushkalkatara/catkin_ws/src/VisualStates/src/visualstates/parsers/filemanager.py", line 79, in open
    rootState.parse(rootNode)
  File "/home/pushkalkatara/catkin_ws/src/VisualStates/src/visualstates/core/state.py", line 167, in parse
    transitionNodes = childState.parse(childNode)
  File "/home/pushkalkatara/catkin_ws/src/VisualStates/src/visualstates/core/state.py", line 179, in parse
    transition.parse(tranNode, statesById)
  File "/home/pushkalkatara/catkin_ws/src/VisualStates/src/visualstates/core/transition.py", line 163, in parse
    self.name = transitionElement.getElementsByTagName('name')[0].childNodes[0].nodeValue
IndexError: list index out of range
[2]    14959 segmentation fault (core dumped)  rosrun visualstates main.py ./src/prius_overtake/prius_overtake.xml
```
Transitions without name should be allowed.

sudo-panda, 2019-04-04:
I would like to work on this issue. 
I think we should check the input of the transition's name and ask the user to input its name again.

pushkalkatara, 2019-04-05:
Hi @sudo-panda .  The transition name can be empty too. While parsing the XML to automata using [transition.py](https://github.com/JdeRobot/VisualStates/blob/master/src/visualstates/core/transition.py), it expects a value but as the node is empty, when it tries to access the node value, it gives `list index out of range` error. You can check if the node is empty and then access the nodeValue.

sudo-panda, 2019-04-05:
I have added a try catch block to rectify the issue. And it is now fixed.

sudo-panda, 2019-04-05:
Can I claim another issue before the pull request is merged?

pushkalkatara, 2019-04-05:
@sudo-panda Thanks for the PR. it solves the issue and I have merged it. 

Yes, you can approach as many as issues you want.

closed 2019-04-05 by pushkalkatara