JdeRobot/VisualStates #93: [good-first-issue] new visualstate should clear the last file from filemanager

okanasik, 2019-03-13:
When we first open a visualstates behavior, then we choose to create a new visualstates behavior, but the filemanager still keeps the filename of the previously opened behavior. Therefore, when the user clicks the save button, the new behavior will override the previously opened behavior.

sudo-panda, 2019-04-06:
I would like to work on this issue.
Do I just need to set filemanager.fullPath to empty when it is opening a new file or am I going about it wrong again?

okanasik, 2019-04-13:
Yes, it could be fine to set it to empty, but you should also check whether the empty filename is checked when user wants to save the file. In that case, file save dialog should open since the there is no filename for the current behavior.
Another important issue about the filemanager is that since behavior import also uses the filemanager, it temporarily sets a filename during the import, but as the import ends, it should reset the previous value of the filemanager file path.
Can you check these two cases?

sudo-panda, 2019-04-13:
Ok I will look into these cases.

sudo-panda, 2019-04-15:
In the case of import behavior, it doesn't set the path permanently so it isn't a problem.

sudo-panda, 2019-04-15:
I have solved to issue but will send a pull request later will more solved issues.

okanasik, 2019-06-02:
Solved by #106

closed 2019-06-02 by okanasik