JdeRobot/VisualStates #95: If empty string in AddWorkspace path, VisualStates unexpectedly exits

pushkalkatara, 2019-03-15:
If empty string is provided in the path, then visualstates shutdowns with the following error -

```
Traceback (most recent call last):
  File "/home/pushkalkatara/catkin/src/VisualStates/src/visualstates/gui/dialogs/workspacetab.py", line 75, in addWorkspace
    if self.isValidWorkspace(newDir):
  File "/home/pushkalkatara/catkin/src/VisualStates/src/visualstates/gui/dialogs/workspacetab.py", line 85, in isValidWorkspace
    files = os.listdir(dir)
OSError: [Errno 2] No such file or directory: ''
[1]    15434 abort (core dumped)  rosrun visualstates main.py
```

sudo-panda, 2019-04-06:
Do we need to show the "This is not a catkin workspace ..." warning or just exit the dialog?

pushkalkatara, 2019-04-06:
Hi @sudo-panda , maybe you can update the text in the dialog with "This is not catkin workspace or path not found".

commit 7eb2d691 (2019-04-06): Transition ID import update, empty workspace directory error fix (#95, #94)

* Add a check for empty workspace directory to fix #95

* Add a function to update transition IDs to fix #94
closed 2019-04-06 by pushkalkatara