JdeRobot/VisualStates #96: If condition is empty in transition, XML fails to parse

pushkalkatara, 2019-03-16:
While developing automata if condition is not given, the XML file saves but while reloading it in VisualStates the following error occures.
```
Traceback (most recent call last):
  File "/home/pushkalkatara/catkin/src/VisualStates/scripts/main.py", line 30, in <module>
    mainWindow.openFile(sys.argv[1])
  File "/home/pushkalkatara/catkin/src/VisualStates/src/visualstates/gui/visualstates.py", line 215, in openFile
    (self.rootState, self.config, self.libraries, self.globalNamespace) = self.fileManager.open(fileName)
  File "/home/pushkalkatara/catkin/src/VisualStates/src/visualstates/parsers/filemanager.py", line 79, in open
    rootState.parse(rootNode)
  File "/home/pushkalkatara/catkin/src/VisualStates/src/visualstates/core/state.py", line 179, in parse
    transition.parse(tranNode, statesById)
  File "/home/pushkalkatara/catkin/src/VisualStates/src/visualstates/core/transition.py", line 162, in parse
    self.setCondition(transitionElement.getElementsByTagName('condition')[0].childNodes[0].nodeValue)
IndexError: list index out of range
```

okanasik, 2019-04-13:
I think we need to generate a more encompassing user input validation so that we can keep all such policies in a single place to have clean code space. I will change the title of the issue to a more general user input validation.

okanasik, 2019-06-02:
@sudo-panda as part of hardening efforts, can you test all user inputs and whether they are validated according to the function. As a humble suggestion, you can first list all user input interfaces of the tool and determine what is the valid input for the considered input interface and finally you can test them. What do you think?

sudo-panda, 2019-06-02:
I will work on it.

sudo-panda, 2019-06-02:
These are my findings:

| Input Name | Input Type | Error(if any) |
|---------------|--------------|---------------|
| File -> Open -> File Dialog | VisualStates Files | Program crashes if any other .xml file is provided |
| File -> Import -> File Dialog | VisualStates Files | Program crashes if any other .xml file is provided |
| Spawn Transition | Two States | It allows the transition to have the origin and destination to be same (Should this be allowed) |
| State Code | Code | |
| Temporal Transition Code -> Time | Integer | If any other data type is used the program crashes  |
| Conditional Transition Code -> Condition | Condition |  **If left empty the file saves but does not open again** |
| Transition Code -> Code | Code | |
| Data -> Timer | Integer | If any other data type is used the program crashes |
| Data -> Global Namespace | Code | Ctrl+G is also allotted to State Namespace and C++ Code Generation |
| Data -> State Namespace | Code | Ctrl+G is also allotted to Global Namespace and C++ Code Generation |
| Actions -> Libraries | Libraries | Should check if added libraries are already present |
| Actions -> ROS Config -> Package | Build and Run dependencies| **Empty topics are saved but when the file is opened it shows an error.** Should check if added dependency already present |
| Actions -> ROS Config -> Topics | Subscribers and Publishers | Should check if Method Name, Variable Name or Topic Name entered is already present |
| Actions -> ROS Config -> Workspaces | Directories | Should check if chosen directory is workspace or not in the dialog itself (may not be possible) and check if added workspace is already.|
| Actions -> C++ Generation | Click | Ctrl+G is also allotted to Global Namespace and State Namespace |
| Action -> Python Generation | Click | |

Also it would be good if python and c++ code highlighter are added.
And instead of keeping the up button it would be better to add a root state in the state lists.

sudo-panda, 2019-06-10:
@okanasik Sir, Spawning transitions allows the transitions to have the origin and destination to be same.
Is this a feature or a bug.

okanasik, 2019-06-17:
Actually, I am not able to come up with a valid use-case for self transitions, but technically finite state machines can have self transitions. It could be better if we disallow that.

okanasik, 2019-06-24:
Since we create specific issues from these user input validation problems, we can close this issue. See the list of created issues here:
- #113 
- #116 
- #117 
- #118 
- #119

closed 2019-06-24 by okanasik