JdeRobot/VisualStates #100: Jderobot logo size is very small in left corner of the tool.

pushkalkatara, 2019-04-06:


okanasik, 2019-04-13:
I made it smaller some time ago to make use of all the space for the good of the user. I checked many tools, I have been using in daily basis and I see that none of them have logo taking that much screen space and I decided to make it smaller. Also, to include the relevant references, I just designed a new about window with a big nice JdeRobot logo.
This was the context. What do you think? Do you think we still need to put the logo somewhere on the active window?

pushkalkatara, 2019-04-14:
Oh right. I checked the `About -> Help` section of the tool. It has the logo as well as links too. Maybe we can remove it or it is just fine.

okanasik, 2019-04-14:
Are you referring to the issue while saying "maybe we can remove it or it is just fine" What is your opinion? You want to have a visible logo? If yes, can you write why do we need that?

pushkalkatara, 2019-04-19:
I was referring to the logo on the active window. As there is an `About -> Help` section in which logo and relevant links are there, I think we won't need the logo on the main page. We can remove it or it is just fine of this size.

okanasik, 2019-05-18:
When I was removing from the main window, I thought it would be good to keep small portion of it. Now, I think we can remove that small logo since it thickens the status bar.

okanasik, 2019-06-02:
Solved by #112

closed 2019-06-02 by okanasik