JdeRobot/VisualStates #99: If XML is imported and not saved, code generation causes the program to crash.

sudo-panda, 2019-04-06:
If all of the XML files are imported and then the resulting XML is not saved generating Python or C++ code crashes the program with error:
```
Traceback (most recent call last):
  File "/home/sudo-panda/Workspaces/jde_robot/src/VisualStates/src/visualstates/gui/visualstates.py", line 313, in generateCppAction
    generator.generate(self.fileManager.getPath(), self.fileManager.getFileName())
  File "/home/sudo-panda/Workspaces/jde_robot/src/VisualStates/src/visualstates/generators/cpprosgenerator.py", line 36, in generate
    os.makedirs(projectPath + os.sep + 'src')
  File "/usr/lib/python2.7/os.py", line 157, in makedirs
    mkdir(name, mode)
OSError: [Errno 13] Permission denied: '/src'
[1]    3073 abort (core dumped)  rosrun visualstates main.py

```
**Possible Fix:**
Automatically save file before generating code

pushkalkatara, 2019-04-06:
I think the issue is due to the location on which you are saving requires super user permissions, also I see that you are importing a C++ based automata which is not supported yet.
`File "/home/sudo-panda/Workspaces/jde_robot/src/VisualStates/src/visualstates/generators/cpprosgenerator.py", line 36, in generate
    os.makedirs(projectPath + os.sep + 'src')`

I think the way to resolve this issue, as the user click on `File -> Save`, the dialog box should directly point to the ros package which is being developed, also the location of this ros package should be where superuser permissions are not required. Maybe we can display an error to resolve the issue.

sudo-panda, 2019-04-06:
I tried it on both Python and C++ code generators

pushkalkatara, 2019-04-06:
Yes, I was also able to reproduce the issue. Thanks for pointing out. 
```
Traceback (most recent call last):
  File "/home/pushkalkatara/temp/catkin_ws/src/VisualStates/src/visualstates/gui/visualstates.py", line 329, in generatePythonAction
    generator.generate(self.fileManager.getPath(), self.fileManager.getFileName())
  File "/home/pushkalkatara/temp/catkin_ws/src/VisualStates/src/visualstates/generators/pythonrosgenerator.py", line 42, in generate
    fp = open(projectPath + os.sep + projectName + '.py', 'w')
IOError: [Errno 13] Permission denied: '/.py'
[1]    21369 abort (core dumped)  rosrun visualstates main.py
```

okanasik, 2019-04-13:
The expected behavior for such cases can be two ways:
1. Prompting user to save the behavior before the code generation by explaining that code generation will create source files in the same directory of the behavior.
2. Create temporary file in some temporary folder such as `.visualstates` or `/tmp/visualstates/`

What do you think? Which one could be a userfriendly approach. I think the option 1 would be clear for the user.

sudo-panda, 2019-04-13:
I too think that the 1st option is better and more user friendly.

sudo-panda, 2019-04-15:
I have added an info dialog that tells user that it is necessary for the behaviour to be saved before code generation and then it saves the behaviour.

okanasik, 2019-06-02:
Solved by #106

closed 2019-06-02 by okanasik