JdeRobot/VisualStates #109: Deleting states doesn't delete the transtitions that point to it.

sudo-panda, 2019-04-30:
The ideal behaviour would be that both transitions that originate from the state and those that point towards the state should be deleted. Right now it only deletes the transitions that originate from it.

okanasik, 2019-05-18:
Dear @sudo-panda, I am not sure there is such a problem. If a state is removed all transitions that originate and all transitions that point to it is removed in method `removeStateItem` If you think there is still a problem that I do not see, can you write the steps to reproduce the problem. https://github.com/JdeRobot/VisualStates/blob/e79dcf84fb3023e17a3352dccb91f4b11b3001b2/src/visualstates/gui/automata/automatascene.py#L191

sudo-panda, 2019-05-18:
I myself can't reproduce the problem now.
But #110 is necessary for the functioning of #108. Nevertheless closing this issue.

closed 2019-05-18 by sudo-panda