JdeRobot/VisualStates #114: Child Automata doesn't start from initial state if parent state ran once.

sudo-panda, 2019-06-09:
This is the same issue as #70 

[image]
Root State 

[image]
Find Wall State

If state Find Wall is activated it starts at Move Forward and ends at the child state Spin Anticlockwise but when the Find Wall state runs again it starts at Spin Anticlockwise instead of Move Forward

okanasik, 2019-07-16:
I have created a fix (#131 ) for this issue, but i am not sure how that will affect previously created behaviors. Dear @sudo-panda  and @pushkalkatara Can you test this new fix with the existing visualstates examples (https://github.com/JdeRobot/VisualStates-examples)

Also, I think there is a design flaw for the behavior depicted above. The states should be designed as much as self-contained as possible. For example, when `Find a Wall` behavior reaches `Spin Anticlockwise`, it expects the state change from the top level state. However, it can also have transition to other states based on the current state. Well, it is not a big issue, but finite state machine based behavior definition is more powerful when states have a single well defined, self-contained behavior.

pushkalkatara, 2019-08-09:
Hi, @okanasik I checked the fix with [prius_overtake](https://github.com/JdeRobot/VisualStates-examples/tree/master/prius_overtake) example with 2 cars to overtake. The behavior worked fine. I think if the feature works with this example, it would work with all examples.

okanasik, 2019-10-04:
fixed by 294e6094e850470ccf42e4b6bdd91c4914f6279d

closed 2019-10-04 by okanasik