JdeRobot/VisualStates #116: Using data types other than integers in Temporal Transition Code crashes the program

sudo-panda, 2019-06-10:


sudo-panda, 2019-06-11:
I would like to work on this issue.

sudo-panda, 2019-06-11:
Can I add support for using parameters in Temporal Transition Code, or should I do that in another pull request.

okanasik, 2019-06-17:
I think it is better to add this feature as part of parameterization.

sudo-panda, 2019-06-17:
I solved it on the colab repo but I am unable to create a pull request due it not being a fork. Any ideas?

okanasik, 2019-06-17:
I have written a method on the hangouts, did you try that? 
> you should have a fork of the visualstates as your personal repository then you should add this as another remote repository to your local-colab repository then you should push your branch that you would like to create pull request from to that forked remote repo finally through github ui you can create pull request from your personal repo I do not know any other way that you can create a pull request

Also do not forget that before working on any single issue, you should checkout the master branch. pull the latest version from the remote repo. create a new branch for the issue/feature make your changes and push your changes to the remote repo. finally create a pull request.

If you still have problems please let me know.

sudo-panda, 2019-06-17:
Sorry I didn't get the notification for the message. I will try that method and if I get merge conflicts while trying pull from the remote repo should I solve it and commit or it is considered a bad practice.

okanasik, 2019-06-17:
If you need to merge because of a conflict that is not related to current changes that you have build for the respective issue, it would be bad practice. However, if you need to merge because your changes overwrites the changes done between your last pull and this pull, it is fine. Also do not hesitate to break things, we can always fix things in git :)

sudo-panda, 2019-06-17:
Alright, thanks

okanasik, 2019-07-09:
fixed by #126

closed 2019-07-09 by okanasik