JdeRobot/VisualStates #117: If childNodes are empty in xml file it does not open

sudo-panda, 2019-06-10:
For Example, If we save a VisualStates file which empty conditional transition code it saves but when opening or importing the file it crashes the program throwing an `IndexError` Exception.

Wherever the inputs can be empty we need to catch the error, in others we need to validate user inputs to check that they aren't empty.

okanasik, 2019-06-17:
I have changed the title of the issue as "XML Validation" because, we should have well-defined XML schema and make sure that XML parsing and creation conforms these norms. In the case of errors, it should be failsafe. However, it should not work by getting everything in try and except clauses.

sudo-panda, 2019-07-03:
I would like to work on this issue.

okanasik, 2019-07-09:
Partially fixed by #128. We need to have complete XML validation. For example, if a user selects any irrelevant XML document to open in the app, the app should show an error message and should not crash.

sudo-panda, 2019-07-09:
This commit 356c893801c08de3bdd7a2cf55c9f222651cbcea does that @okanasik

okanasik, 2019-07-09:
Yes, you are right. Sorry I missed that, but the file is not a valid XML file (i.e. not conforming to the general XML schema) the app crashes. Actually, the XML parser gives an error of type `xml.parsers.expat.ExpatError`  Maybe we can solve this problem by using try catch mechanism to catch only this type errors. What do you think?

sudo-panda, 2019-07-09:
Yes that can be implemented.

okanasik, 2019-07-16:
Invalid XML file error is handled with the commit 700d24e02427dbad909801c986ee178a34cb44bc

closed 2019-07-16 by okanasik