JdeRobot/VisualStates #120: A tabbed design of VisualStates.

sudo-panda, 2019-06-10:
Using dialog for every type of input is not user friendly.

Hence I suggest a tabbed design which might loo somewhat like these pictures:
[image]
[image]
[image]

Of course this is a rough estimate and the more inputs received the better would be the design.

sudo-panda, 2019-06-10:
I haven't shown the where the user will be able to interact with the parameters list. I think it would be better if the parameter list would be displayed beside the code in a collapsible sidebar

pushkalkatara, 2019-06-15:
Hi @sudo-panda . Could you please elaborate the details of the tab. I can see 'Root', 'State 1', 'State 1' and 'Transition 1'. Does this mean source code for 'State 1' and 'Transition 1'? Also what does the 'Root' and 'State 1' tab depicts?

sudo-panda, 2019-06-15:
The tabs with </> sign mean that it shows the source code for the state or transition
While ones with a blue circle and red square show the canvas where the child 
 states and transitions of the state mentioned is shown.

pushkalkatara, 2019-06-15:
Okay great. the design looks good. Could you please explain why `Root` and `State1` with blue circle and red square are two separate tabs?

okanasik, 2019-06-17:
- When building a behavior, you would like to write code by checking the code of the other states. Therefore, we let the user to access all possible code locations via such tabbed design. The life of the programmer would be easier.
- However, the user should be able to open these tabs via right click menu and should be able to close them as needed.

pushkalkatara, 2019-08-10:
Hi all, what is the status of this issue?