JdeRobot/VisualStates #134: Error while opening a xml file in visualstates.

pankhurivanjani, 2019-08-24:
While working on Visualstates with ROS2, I got the error of 

``` IndexError: list index out of range```

When I checked it with ROS1, I am getting same error here too, I didn't change anything in the code for ROS1. I am using ROS Melodic with Ubuntu 18.04.

This error is coming when I try to open .xml file in Visualstates GUI.

Exact error is:

```pankhuri@pankhuri-LENOVO-IDEAPAD-500-15ISK:~/catkin_ws$ rosrun visualstates main.py 
Gtk-Message: 12:05:57.318: GtkDialog mapped without a transient parent. This is discouraged.
Traceback (most recent call last):
  File "/home/pankhuri/catkin_ws/src/VisualStates/src/visualstates/gui/visualstates.py", line 205, in openAction
    self.openFile(fileDialog.selectedFiles()[0])
  File "/home/pankhuri/catkin_ws/src/VisualStates/src/visualstates/gui/visualstates.py", line 208, in openFile
    (rootState, config, libraries, globalNamespace) = self.fileManager.open(fileName)
  File "/home/pankhuri/catkin_ws/src/VisualStates/src/visualstates/parsers/filemanager.py", line 80, in open
    globalNamespaceNode = doc.getElementsByTagName('VisualStates')[0].getElementsByTagName('global_namespace')[0]
IndexError: list index out of range
Aborted (core dumped)
```

sudo-panda, 2019-08-24:
Can you send the XML too?

pankhurivanjani, 2019-08-24:
Here, it is https://github.com/JdeRobot/base/tree/master/src/examples/visualStates/ros/obstacleAvoidancePython

pankhurivanjani, 2019-08-24:
Instead of examples given in jderobot base repository, I used examples from visualstates-example repository and it worked fine.

closed 2019-08-24 by pankhurivanjani