JdeRobot/VisualStates #136: Issue with running kobuki obstacle avoidance in python with ROS1

pankhurivanjani, 2019-08-25:
I am getting errors in running this example with visualstates

My System Info:
OS: ubunutu 18.04
ROS: Melodic
Python: In this case, python 2 itself

```pankhuri@pankhuri-LENOVO-IDEAPAD-500-15ISK:~/catkin_ws$ rosrun kobuki_obstacle_avoidance kobuki_obstacle_avoidance.py --displaygui=true
Unable to register with master node [http://localhost:11311]: master may not be running yet. Will keep trying.
runtime gui enabled
WARNING: QApplication was not created in the main() thread.
Exception in thread Thread-7:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/pankhuri/catkin_ws/src/VisualStates/src/visualstates/codegen/python/state.py", line 97, in run
    if tran.checkCondition():
  File "/home/pankhuri/catkin_ws/src/kobuki_obstacle_avoidance/kobuki_obstacle_avoidance.py", line 104, in checkCondition
    self.namespace.calculate_obstacle()
AttributeError: Namespace0 instance has no attribute 'calculate_obstacle'
```

sudo-panda, 2019-08-25:
Its a VisualStates-examples issue so could you open it there?

jmplaza, 2019-08-26:
sure, @pankhurivanjani has just created [it there](https://github.com/JdeRobot/VisualStates-examples/issues/22)

closed 2019-08-26 by jmplaza