JdeRobot/VisualStates #140: New webpage for VisualStates based on Minimal-Mistakes template

jmplaza, 2019-10-06:
It will be available at[ this URL](https://jderobot.github.io/VisualStates) and its source code at the [github repository](https://github.com/JdeRobot/VisualStates/tree/master/docs) of VisualStates

jmplaza, 2019-10-14:
It still does not work properly @pushkalkatara 

From the current contents in the repo I can't deploy it locally, and it also does not work on the GitHub web either. Following [these instructions](https://github.com/JdeRobot/jderobot.github.io/blob/master/README.md) the local deployment works for JdeRobot webpage and for DetectionSuite webpage, but not for VisualStates webpage....

Does VisualStates webpage work for you with a similar appearance of DetectionSuite and JdeRobot organization?

pushkalkatara, 2019-10-25:
I recently made a new PR and merged it into the master. I made changes to the docs and its working now. Please have a look here at https://jderobot.github.io/VisualStates/ . I ported all the documentation to the new theme. Further, we can discuss and make commits to improve upon it.

okanasik, 2019-10-26:
@pushkalkatara The image links are broken. Also, we had a vectoral drawing version of JdeRobot in resources directory. Can you this image for the home page? Also white text color on white background is hard to read and does not look nice. Can you give different color to these texts or background.

pushkalkatara, 2019-11-02:
@okanasik I fixed the backgrounds and the text color with recent commits. I am facing some weird behavior, like the [usage guide](https://jderobot.github.io/VisualStates/features/usage) links are working locally but not when I push on the website. Also, locally, the title is VisualStates but it changes to `what is VisualStates` on the server.