JdeRobot/VisualStates #146: VisualStates Release for ROS Melodic

jiban-jyoti, 2020-01-09:
What is the release date for VisualStates package for ROS-melodic as currently I am doing source installaton to run the exercises based on VisualStates. Also well done for maintaining this package and I am excited to know the release dates.

okanasik, 2020-01-13:
Thank you for your interest in the project. We are ready for the release, we are planning to make the release in the following week. I will let you know when we release. If you encounter any issues please do not hesitate to report.

jiban-jyoti, 2020-01-19:
Hello @okanasik I think the visualstates package for ROS-melodic is released now. Can you tell me the instructions to install it. I tried source installation but I am getting errors.

rosrun visualstates main.py
Traceback (most recent call last):
  File "/home/jiban/catkin_ws/src/VisualStates/scripts/main.py", line 25, in <module>
    from visualstates.gui.visualstates import VisualStates
ImportError: No module named visualstates.gui.visualstates

okanasik, 2020-01-24:
Hi @jiban-jyoti , I checked the problem. If you rename the directory to all lowercase letters, it should work. Did you solve the problem?

jiban-jyoti, 2020-01-28:
Yes Sir, It is working perfectly now. We should update the commands also.

closed 2020-01-29 by jiban-jyoti