JdeRobot/VisualStates #152: Import error in running binary package of visualstates

pankhurivanjani, 2020-03-20:
Hi I am trying to run the newer ros-melodic release of visualstates. 

When I run the file after sourcing the environment, I get this import error. 

```
pankhuri@pankhuri-LENOVO-IDEAPAD-500-15ISK:~$ rosrun visualstates main.py 
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/visualstates/main.py", line 25, in <module>
    from visualstates.gui.visualstates import VisualStates
  File "/opt/ros/melodic/lib/python2.7/dist-packages/visualstates/gui/visualstates.py", line 28, in <module>
    from .dialogs.librariesdialog import LibrariesDialog
  File "/opt/ros/melodic/lib/python2.7/dist-packages/visualstates/gui/dialogs/librariesdialog.py", line 26, in <module>
    from visualstates.gui.util.editablestringwidget import EditableStringWidget
ImportError: No module named util.editablestringwidget
```

@pushkalkatara could you suggest something for this?

jiban-jyoti, 2020-03-20:
@pankhurivanjani I am also facing the same issue while running the new release of visualstates.

pushkalkatara, 2020-03-21:
Hi @pankhurivanjani @jiban-jyoti Thanks for pointing out the issue. Please follow the following steps while we are trying to resolve the issue:

In your ROS workspace,

```
cd src
git clone https://github.com/JdeRobot/VisualStates.git
cd ..
catkin_make
source devel/setup.bash
rosrun visualstates main.py
```

closed 2020-04-22 by pankhurivanjani