JdeRobot/VisualStates #153: Permission Error in generating python node of examples with ros package

pankhurivanjani, 2020-04-22:
While working with recently released ros package when I try to generate python nodes from visualstates I get the permission error.

```
pankhuri@pankhuri-LENOVO-IDEAPAD-500-15ISK:~/catkin_ws/src/kobuki_obstacle_avoidance$ rosrun visualstates main.py kobuki_obstacle_avoidance.xml 
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/visualstates/gui/visualstates.py", line 331, in generatePythonAction
    generator.generate(self.fileManager.getPath(), self.fileManager.getFileName())
  File "/opt/ros/melodic/lib/python2.7/dist-packages/visualstates/generators/pythonrosgenerator.py", line 42, in generate
    fp = open(projectPath + os.sep + projectName + '.py', 'w')
IOError: [Errno 13] Permission denied: '/.py'
Aborted (core dumped)

```
Problematic file: [line-43_python](https://github.com/JdeRobot/VisualStates/blob/master/src/visualstates/generators/pythonrosgenerator.py#L42)

Testing with cop node generation also resulted in permission issues:

```
pankhuri@pankhuri-LENOVO-IDEAPAD-500-15ISK:~/catkin_ws/src/kobuki_obstacle_avoidance$ rosrun visualstates main.py kobuki_obstacle_avoidance.xml 

Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/visualstates/gui/visualstates.py", line 315, in generateCppAction
    generator.generate(self.fileManager.getPath(), self.fileManager.getFileName())
  File "/opt/ros/melodic/lib/python2.7/dist-packages/visualstates/generators/cpprosgenerator.py", line 36, in generate
    os.makedirs(projectPath + os.sep + 'src')
  File "/usr/lib/python2.7/os.py", line 157, in makedirs
    mkdir(name, mode)
OSError: [Errno 13] Permission denied: '/src'
Aborted (core dumped)
```

Problematic file: [line-36_cpp](https://github.com/JdeRobot/VisualStates/blob/master/src/visualstates/generators/cpprosgenerator.py#L36)

This error is while working with visualstates melodic ros package. When I did the same stuff using source installation all went well!

pushkalkatara, 2020-04-23:
Hi @pankhurivanjani 

Did you save the XML in your catkin workspace before generating the python node. I think VisualStates doesn't know location of your catkin workspace. Once you save the XML in catkin workspace, it would generate the Python node there.

pankhurivanjani, 2020-04-24:
Yes, it  is working  fine now. 
Thanks.

closed 2020-04-24 by pankhurivanjani